dmadison/ServoInput #12: board MKR WiFi 1010 not supported

AdrianTschannen, 2020-10-27:
Hi
Thank you very much for your effort to develop this library!
It seems, that MKR boards are not supported. As mentioned in the error message, I ask you to add this board (or all MKR boards).
Thank you very much! Adrian

Error message from arduino compiler:

"In file included from C:\Users\adria\OneDrive\Dokumente\Arduino\libraries\ServoInput-1.0.0\src/ServoInput.h:27:0,

                 from C:\Users\adria\OneDrive\Dokumente\Arduino\libraries\ServoInput-1.0.0\examples\BasicAngle\BasicAngle.ino:29:

C:\Users\adria\OneDrive\Dokumente\Arduino\libraries\ServoInput-1.0.0\src/ServoInput_Platforms.h:44:2: error: #error "The ServoInput library does not support this board (platform). Please create a feature request here: https://github.com/dmadison/ServoInput/"

 #error "The ServoInput library does not support this board (platform). Please create a feature request here: https://github.com/dmadison/ServoInput/"

  ^~~~~

Multiple libraries were found for "ServoInput.h"
 Used: C:\Users\adria\OneDrive\Dokumente\Arduino\libraries\ServoInput-1.0.0
exit status 1
Error compiling for board Arduino MKR WiFi 1010."

dmadison, 2020-10-28:
Hi there! Thanks for making the feature request.

I've pushed a few commits to the [`boards-mkr` branch](https://github.com/dmadison/ServoInput/tree/boards-mkr) that should add support for the Arduino MKR boards, including the MKR WiFi 1010. This compiles on my machine but I do not have any MKR boards on hand to test with.

Please test that branch with your hardware when you get a chance.

AdrianTschannen, 2020-10-28:
Hi
I've tested it and: **it works perfectly!** Thank you very much for this. Now, I will add some more libraries into my code and hope, the interrupts don't influence each other...
Two questions:
- Will it be possible that you implement the MKR capability in the ordinary version of your library or will it be necessary to use the specific MKR version in the future?
- I will probably also use a NANO 33 BLE SENSE board for a project. Is it possible, to integrate this board also into the ServoInput library?
Thank you very much, Adrian

dmadison, 2020-10-28:
Thanks Adrian. While I have you, could you please test all of the interrupt capable pins on the board with the library? It looks like the following pins support interrupts:

```
A1, A2, D0, D1, D4, D5, D6, D7, D8, D9
```

You would need to move your signal pin connection and update the pin number (`ServoInputPin<#>`) in the code. Any example will do, I just want to make sure the interrupt logic is working properly.

>Will it be possible that you implement the MKR capability in the ordinary version of your library or will it be necessary to use the specific MKR version in the future?

If you can confirm that the `boards-mkr` branch works correctly on your hardware for all pins, I'll merge the changes into master.

>I will probably also use a NANO 33 BLE SENSE board for a project. Is it possible, to integrate this board also into the ServoInput library?

Please create a separate issue for that. That board runs a RTOS so it might be trickier to get working with how the library is written.

AdrianTschannen, 2020-10-28:
Hi
Test shows following results:
Pins D0, D1, D4, D5 do work correctly.
Pins A1, A2, D6, D7, D8, D9 don't work.
Adrian

dmadison, 2020-10-28:
Thanks for testing Adrian. I've updated the branch with one more change. Please update your library and test one more time.

AdrianTschannen, 2020-10-28:
OK, I've done the test again. Unfortunately with the same result (4 pins work / rest not). Probably there is one possible issue on my test-side. As I have to use level shifting 3.3 V, I did the test using a "MKR connector carrier". This board provides grove adapters. The connection from the MKR board is not that clear. Probably I have to use the pins from the board directly. I've ordered separate level shifter and will redo the test - hopefully within the next days. Adrian

AdrianTschannen, 2020-10-30:
I've now tested all pins using the signal directly from the MKR board. Thank you for your test program. Result: All pins are working! The "MKR connector carrier" was the issue: The connection from the Arduino to the grove plug is not that clear.
If you don't mind: Please integrate the MKR funcionality in the original version.

you are helping me a lot! Thanks!!!

PR #14 (2020-10-31): Arduino MKR Boards Support
Adds support for the Arduino MKR boards which use the SAMD21 processor. Closes #12.
closed 2020-10-31 by dmadison