dmadison/ServoInput #19: Will this work with current ESP32?

KnightDeluxe12, 2022-03-23:
I'm wondering if this library is also applicable for esp32? because i broke my nano and i only have esp32 as a spare for a rc car project thanks

dmadison, 2022-03-23:
It should, although I've never tested it myself.

bisak, 2022-08-14:
@dmadison Unfortunately I cannot get any of the examples to work. Always stuck at `Waiting for servo signals...`. Whereas I'm certain connections are OK because I'm able to get some output via a simple pulseIn(). 
How could we go about debugging this?

dmadison, 2022-08-15:
What board are you using?

If you skip the `available()` check and call `getPulseRaw()`, do you get any values?

If not then it's likely an interrupt issue of some sort. The first thing I would test is that the interrupt routine is being triggered without the library. Something like:

```cpp
volatile bool triggered = false;

void ICACHE_RAM_ATTR isr() {
    triggered = true;
}

attachInterrupt(pin, isr, CHANGE);
while(true) {
    if(triggered) {
        triggered = false;
        Serial.print("T ");
        Serial.println(micros());
    }
}
```

If that works, expand it to call the public `isr()` function of the template class from within your own ISR. If *that* works then it's likely to be an issue with the automatic ISR setup, otherwise it may be an issue with the timestamp functions.

SkylarTheGrey, 2024-01-24:
I was able to get @clowrey pass-through ISR to work. 
Can also confirm that on an ESP-WROOM-32 that your code with the declaration in setup() does work, though of course its then not usable in other functions.

dmadison, 2024-01-29:
Thank you everyone who has responded with information about this issue. Hopefully this should be fixed with the latest release and the changes from PR #30. The interrupt handler must now be attached by the user, which avoids the automatic attachment pitfalls present in the ESP32 core.

Please test [the 2.0.0 release](https://github.com/dmadison/ServoInput/releases/tag/v2.0.0), and feel free to reopen this issue if anyone still has problems.

closed 2024-01-29 by dmadison