dmadison/ServoInput #21: Couple of Comments not issues

EclipseBoom, 2022-08-18:
First thanks for making this.  It looks really clean.

1) Works with ATTiny1626 and the megaTinyCore core for the newer ATTiny CPU chips (which are really superior to the ancient ATMega168/328 chips)

2) In your examples you assume the signal range is symmetrical around the average of the max and min values.  This is not always true especially for pistol style RC car radios.  I always store the center point along with the max and min for this situation.

3) When converting from pulse time to output value I always have a deadband around zero so the output isn't' jittering slightly above and below zero constantly.  Then ramp linearly or exponentially to the max/min from the edge of the deadband.  I realize this is extra code but you have most of it already done with the map function.

Again thanks for putting this together.  Much appreciated.

Chris

dmadison, 2022-08-18:
Hi Chris, thanks for your comments.

The reason there's not more complex functionality for interpreting the received data is because the signal itself contains no metadata, it's purely a time period between two extremes. More advanced functionality for rescaling and remapping is left to the user for simplicity's sake. It would be possible to add a rescaling function callback, but I'm not sure if that's really necessary since the mapping functions (sans deadzone) are more-or-less variations of the same.

The deadzone functionality was added for my own purposes, and in hindsight probably shouldn't have been included with this (servos don't really have "deadzones" in that sense). If you need it I have [another small library](https://github.com/dmadison/CtrlUtil) with a deadzone function that can zero out the bottom, middle, or top of an input range. That library is in need of some polishing but it will do the trick.

EclipseBoom, 2022-08-18:
No problem.  I've worked it out in the past.  Just thought anyone making an actual system instead of a classroom example would need something like that.  Makes sense to leave implementation details up to the user rather than try and cover every possible scenario in the library.

Feel free to close this "issue" if you like.

closed 2022-08-18 by dmadison