dmadison/ServoInput #23: Trouble with more than two servo inputs

savall21, 2022-12-29:
I have three servo inputs connected. Each one works individually. However, when all three are connected one of them always fails to work properly.  I have three switch channels connected from channels 4, 5 & 6.  Chanel six is a momentary switch, channel 5 is three position and channel four is a two position.  Channels 4 and 6 work flawlessly. If I plug in channel 5 then it only reads position 2 & 3. Also, when channel 5 is plugged in channel 6 works intermittently.  

I'm using an Arduino Nano Every with Ch 6 on pin 12, Ch 5 on pin 8 and Ch 4 on pin 10. The ground is connected to pin 29.
Code:

`/*
 *  Project     Servo Input Library
 *  @author     David Madison
 *  @link       github.com/dmadison/ServoInput
 *  @license    LGPLv3 - Copyright (c) 2020 David Madison
 *
 *  This example is part of the Arduino Servo Input Library.
 *
 *  This program is free software: you can redistribute it and/or modify
 *  it under the terms of the GNU Lesser General Public License as published by
 *  the Free Software Foundation, either version 3 of the License, or
 *  (at your option) any later version.
 *
 *  This program is distributed in the hope that it will be useful,
 *  but WITHOUT ANY WARRANTY; without even the implied warranty of
 *  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 *  GNU Lesser General Public License for more details.
 *
 *  You should have received a copy of the GNU Lesser General Public License
 *  along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 *  Example:      BasicAngle
 *  Description:  Reads the PWM signal from a servo motor, converts it to
 *                an angle, and prints that angle over Serial.
 *
 *  Wiring:       Servo signal to pin 2
 */

/* The signal pin for ServoInput MUST be an interrupt-capable pin!
 *     Uno, Nano, Mini (328P): 2, 3
 *     Micro, Leonardo (32U4): 0, 1, 2, 3, 7
 *             Mega, Mega2560: 2, 3, 18, 19, 20, 21
 * Reference: https://www.arduino.cc/reference/en/language/functions/external-interrupts/attachinterrupt/
 */

#include <SoftwareSerial.h>
#include <ServoInput.h>
//Setup mappings for receiver channels- each channel is mapped to a channel on the transmitter
ServoInputPin<12> ch6; //momentary switch
ServoInputPin<8> ch5;  //three position switch
ServoInputPin<10> ch4;  //two position switch
const int MT = 2;
const int Three = 3; 
int sound_clip = 1;

SoftwareSerial mySerial(2, 1); // RX, TX... Nano Every Tx=1  Rx=2

void sendCmd(int cmd, int lb, int hb, bool reply = false)
{ // standard format for module command stream
uint8_t buf[] = {0x7E, 0xFF, 0x00, 0x00, 0x00, 0x00, 0x00, 0xEF};
int16_t chksum = 0;
int idx = 3; // position of first command byte

buf[idx++] = (uint8_t)cmd; // inject command byte in buffer
if (reply) buf[idx++] = 0x01;
else buf[idx++] = 0x00;// set if reply is needed/wanted
if (hb >= 0) // if there is a high byte data field
buf[idx++] = (uint8_t)hb; // add it to the buffer
if (lb >= 0) // if there is a low byte data field
buf[idx++] = (uint8_t)lb; // add it to the buffer
buf[2] = idx - 1; // inject command length into buffer
buf[idx++] = 0xEF; // place end-of-command byte

mySerial.write(buf, idx); // send the command to module
// for (int i = 0; i < idx; i++) // send command as hex string to MCU
// Serial.printf("%02X ", buf[i]);
// Serial.println();                                                                                                                                                                                                                                                                                                   b
}

void setup()
{

Serial.begin(115200); //for console debugging
delay(2000); //Allow Serial to start up

Serial.println("starting ");
mySerial.begin(9600); //communicate with DF Player
delay(3000); //Allow SoftwareSerial to start up

Serial.println("Stop anything playing"); //in case anything is already playing on reset
sendCmd(0x0E, 0, 0, false);
delay(200); //give a delay for the DF Player to execute the command

//Serial.println("Now volume ");
sendCmd(6, 30, 0, false); //command code can be in decimal too.

//Serial.println("Starting Player");
//sendCmd(0x01, 0, 0, false); // Send a "next track" command that turns play mode on
delay(1000);
Serial.println("_____________________________________");

}

void loop()
{
 //Momentary Switch Channel 6 HS - Play sound
int channel6 = ch6.map(1, MT);
//Serial.println(channel6); //Debugging to read the channel

if (channel6 == 2){
  Serial.print("Sound Activated");
  delay(300);
  play_sound();
}

//Three position switch Channel 5 SE - Select up scanner sound; middle no sound; down turbo boost
int channel5 = ch5.map(1, Three);
//Serial.println(channel5); //Debugging line for channel5
 if (channel5 == 3){
  delay(500);
  Serial.println("Turbo Boost");
  sound_clip = 6;
  }

if (channel5 == 2){
  delay(500);
  Serial.println("Scanner");
  sound_clip = 4;
  }

//Two position switch Chanel 4 SF - Select Sound
int channel4 = ch4.map(1, MT);
if (channel4 == 2){
  delay(500);
  sound_clip = sound_clip + 1; //Increase the number of the selected sound file up to 16 and then reset to 1
    if (sound_clip > 16){
      sound_clip = 1;
    }
  Serial.println("Sound select change");
  Serial.print("Selected sound:");
  Serial.println(sound_clip);
  delay(300);
  }
}

void play_sound() //Subroutine to play the selected sound file
{
  Serial.println("Play sound ");
  Serial.print(sound_clip);
  sendCmd(0x0F, sound_clip, 1, false); //play selected track in folder 01
  delay(200);
  Serial.println(" ");
}
`

Its a great library though! Any help is appreciated.

dmadison, 2022-12-29:
Hello, sorry to hear you're having issues.

What does your wiring setup look like? Have you tried using different pins for the channel inputs?

I see that you're using software serial. Do you have the same issues if you remove the software serial code?

savall21, 2022-12-30:
I'm using basic jumpers on a breadboard.  I tried swapping the ones on the receiver out with no change.  I need software serial to control the DFPlayer Mini.

dmadison, 2022-12-30:
>I need software serial to control the DFPlayer Mini.

I understand that, but software serial may be interfering with the library. For debugging purposes do you have the same issues if you remove the software serial library from the code?

savall21, 2022-12-30:
I swapped the channel and the switch.  It's working now.  Thanks for the help!

closed 2022-12-30 by savall21