dmadison/ServoInput #24: Support for Due

EricB7, 2023-02-26:
Hello David,

This project is exactly what I am looking for, but I am developing with a Due.  Would it be possible to add support for this platform?

Thanks,

Eric

dmadison, 2023-02-26:
Hi Eric. I don't have a Due board on hand but the library compiles for me. Are you having issues?

EricB7, 2023-02-26:
Hi David.  Thanks for the quick reply.  I compiled and downloaded your calibration example.  That goes fine.  No errors.  But I can't get out of the while loop where its waiting for servo.available() to go true.  I've got my an RC receiver channel wired into Due pin 2, and I can see a 3.2v square wave on the pin with my o-scope.  Based on what I read here, I thought maybe the header file needs to be modified to support the Due:

https://www.partsnotincluded.com/servo-input-library-for-arduino/

Thoughts?

Eric

dmadison, 2023-02-26:
Since I wrote that article I've updated the library so that it supports all platforms using the standard `digitalRead()` function, which is slower than direct port access but portable for all platforms.

My hunch is that it's an issue with the ISR attach/detach functions being called in the constructor. Try declaring the object within the `setup()` function and see if it gets past it then:

```cpp
#include <ServoInput.h>

void setup() {
    ServoInputPin<2> servo;

    Serial.begin(115200);
    Serial.println("Starting...");
    while (!servo.available());
    Serial.println("Servo is available");
}

void loop() {}
```

EricB7, 2023-02-27:
OK, I made that change, but the behavior is the same.

dmadison, 2023-02-27:
Hmm. Try creating a generic interrupt function to wrap the object ISR:

```cpp
#include <ServoInput.h>

#define PIN 2
ServoInputPin<PIN> servo;

void isr() {
    servo.isr();
}

void setup() {
    attachInterrupt(digitalPinToInterrupt(PIN), isr, CHANGE);

    Serial.begin(115200);
    Serial.println("Starting...");
    while (!servo.available());
    Serial.println("Servo is available");
}

void loop() {}
```

If that doesn't work there may be an issue calling the timer function from within the ISR.

EricB7, 2023-03-04:
Hi David.  I solved the problem.  I had to add a pulldown resistor to the inputs that are reading the RC servo channels.  You library is working fine as written.  Thanks for helping me with this, and for writing this software.

Regards,

Eric

closed 2023-03-04 by EricB7

dmadison, 2023-03-06:
That's strange, the library should enable the board pull-ups. Ah well, glad you got it working and thanks for following up. 😄