dmadison/ServoInput #27: Compilation Issue on Arduino Nano ESP32

Mattglas, 2023-11-29:
Hello, 

I'm having a strange issue while compiling this library for a Nano ESP32. It looks like the arguments for attachInterrupt were changed for this board. Is there a possible workaround for this?

Console output:
> Arduino\libraries\ServoInput\src/ServoInput.h:99:19: error: macro "attachInterrupt" requires 3 arguments, but only 1 given
>  attachInterrupt();
>                  ^
>Arduino\libraries\ServoInput\src/ServoInput.h:106:23: error: macro "attachInterrupt" requires 3 arguments, but only 1 given
>void attachInterrupt() {
>                       ^
> Exit status 1

Thank you, 
Matthew

dmadison, 2023-12-21:
Hi Matthew, and apologies for the late response. Could you tell me what core files you're using? It looks as though the `attachInterrupt` function is implemented as a macro and not a function, which is confusing the compiler.

dmadison, 2024-01-29:
I haven't heard back from you, so I'm afraid without more information there's not much I can do to debug this issue. Nevertheless I've pushed some changes (PR #30) that should hopefully resolve the problem.

Please test [the 2.0.0 release](https://github.com/dmadison/ServoInput/releases/tag/v2.0.0), and feel free to reopen this issue if you still have problems.

closed 2024-01-29 by dmadison