dmadison/ServoInput #31: Failsafe detection

Typ17b, 2024-05-29:
Hi,
Is it possible, to detect the absence of a valid receiver signal within a given resolution? This way it would be possible to realize a Failsafe detection.

Regards
Stefan

dmadison, 2024-05-29:
Hi Stefan! Could you clarify what "within a given resolution" means?

dmadison, 2024-05-29:
I think I'm understanding you, feel free to correct me if I'm not.

If the RC receiver is still connected to the board but the RC controller isn't sending it signals, it's up to the RC receiver how to handle that. Some receivers will repeat the last known position (pulse duration), some will send a default position, some will stop sending pulses entirely.

If the controller isn't sending signals and the RC receiver **is** still sending pulses (whatever they are), there is nothing the library can do one way or the other. It just sees the pulses, and as far as it knows the pulses are valid.

If the controller isn't sending signals and the RC receiver **is not** sending pulses, the library will report the last known good value it received. If it hasn't ever received a valid signal the initial value is at the bottom of the range (500 in your example) because, well, we have no idea what the initial position is. Some motors (e.g. steering) are 'idle' at center, some motors (e.g. throttle) are 'idle' at low. There's no way of knowing.

The library *does* have a feature to check if there is a new signal, which is the `available()` function. If your receivers are of the latter type where they shut off signals when the RC controller drops out of range, you can check if a signal is `available()` before acting on it, rather than reading continuously:

```cpp
if(servo.available()) {
    // do something with the data
}
```

Does that help?

Typ17b, 2024-05-30:
Hi Dave, you are 100% correct. Some Receivers keep the last known signal, some make weird stuff and some do nothing. Its impossible to check if the signal is valid for the first two cases. Fortunately i must handle the third case, so i will check available() when i am back home later. I will report if it suits my needs.Thanks again!Stefan

dmadison, 2024-08-20:
Hi Stefan, following up on this. Were you able to solve your problem?

Typ17b, 2024-08-20:
Hi Dave,
this solved my problem. Sorry, i forgot to inform you.
I am new to this github stuff, do i have to mark something as solved?

Thanks again for the perfect support!

dmadison, 2024-08-20:
If your problem is solved you can close the issue as completed. I can do that for you.

Thanks for following up!

closed 2024-08-20 by dmadison