dmadison/ServoInput #33: Incompatibility between ServoInput and the Advanced DAC feature of Arduino GIGA R1 WiFi

npf, 2025-01-03:
Using ServoInput 2.0.0 on Arduino GIGA R1 WiFi:

After calling [AdvancedDAC.begin](https://github.com/arduino-libraries/Arduino_AdvancedAnalog/blob/main/docs/api.md#advanceddacbegin), `ServoInput.getPulseRaw()` returns only 0, -1 or 1. It is working just fine before the call.

Advance DAC on Giga R1 is described here:   
https://docs.arduino.cc/tutorials/giga-r1-wifi/giga-audio/

I'm guessing a confict in the usage of interrupts?

dmadison, 2025-01-04:
Hi! Thanks for opening this issue, please provide a little more information:

* Does the `getPulseRaw()` result change over time, or does it stay at 0/-1/1?
* Do you have the same results if you call the DAC begin before the ServoInput begin?
* Which pins are you testing with?
* What signal is driving the `ServoInput` pin?
* Do you have an external pull-up / pull-down resistor on the `ServoInput` pin?

If you could provide a minimal reproducible example sketch that would be helpful.

npf, 2025-01-04:
Hello, thx for the very quick reply!

>     * Does the `getPulseRaw()` result change over time, or does it stay at 0/-1/1?

Yes, it varies.

>     * Do you have the same results if you call the DAC `begin()` before the ServoInput `attach()`?

Yes.
 
>     * Which pins are you testing with?

9, 10, 11, 12, but also 47, 49, 51, 53.
 
>     * What signal is driving the `ServoInput` pin?

Radiolink receiver R6DS / controller AT10 II
 
>     * Do you have an external pull-up / pull-down resistor on the `ServoInput` pin?

nope...

> If you could provide a minimal reproducible example sketch that would be helpful.

https://github.com/npf/ServoInput/commit/995e4dd986c841c66fd3657ecc884b9323cb23ca

Not so minimal, sorry... DAC/USB part from https://docs.arduino.cc/tutorials/giga-r1-wifi/giga-audio/#audio-playback

Best regards

dmadison, 2025-01-04:
Thanks for the follow-up. Let's try to simplify that code:

```cpp
#include <Arduino_AdvancedAnalog.h>
#include <ServoInput.h>

const bool AlwaysPrint = false;

AdvancedDAC dac0(A13);
ServoInputPin<9> inputServo;

unsigned long pulse = 0;

void setup() {
  Serial.begin(115200);
  while (!Serial);

  if (!dac0.begin(AN_RESOLUTION_12, 8000, 32, 64)) {
    Serial.println("Failed to start DAC1 !");
    return;
  }

  inputServo.attach();
}

void loop() {
  if(AlwaysPrint || inputServo.available()) {
      pulse = inputServo.getPulseRaw();
      Serial.println(pulse);
  }
}
```

That compiles on my machine, but I don't have a Giga R1 board on hand to test the hardware. Try it with `AlwaysPrint` both `false` and `true` and please share the results.

If you have access to a scope, it would also be useful to record the servo input pin to see if there is any noise.

npf, 2025-01-06:
> Thanks for the follow-up. Let's try to simplify that code:
> That compiles on my machine, but I don't have a Giga R1 board on hand to test the hardware. Try it with `AlwaysPrint` both `false` and `true` and please share the results.

Thanks for the code!
* With `AlwaysPrint=false`, there's nothing on the serial console (I then added `print("."); delay(100);` in `loop()`, dots are displayed alone).
* With `AlwaysPrint=true`, I randomly get either `0`, `1`, or `4294967295`.

> If you have access to a scope, it would also be useful to record the servo input pin to see if there is any noise.

I changed my setup to use an Arduino Uno to feed the PWM signal (I do not have access to the Radiolink RC controller/receiver anymore right now).

Here is the oscilloscope output, as well as a [video](https://photos.app.goo.gl/Dumu7zK4PAYMYwvK7) of my setup.
[image]

npf, 2025-01-06:
Btw, initially, I ran the DAC code on the M4 core, and ServoInput (and more) on the M7 core, with a [RPC](https://docs.arduino.cc/tutorials/giga-r1-wifi/giga-dual-core/#remote-procedure-call-rpc) call from M7 to M4 to enable the DAC/sound play. The issue arised, so I then moved everything on the M7 only to try and narrow down.

dmadison, 2025-01-07:
Hmm.

What's strange is the `4294967295` result. That would indicate that the `micros()` call returned a number prior to the previous recorded one, which shouldn't happen (`micros()` *should* be monotonic).

How quickly do the numbers change from 0/1/4294967295? Is it every function call, or is it aligned with the falling edge of the signal? I'm trying to figure out if it's an issue with the contents of the interrupt or the interrupt itself.

---

Try this, both with and without the `dac0.begin()` line:

```cpp
#include <Arduino_AdvancedAnalog.h>

AdvancedDAC dac0(A13);
const int ServoPin = 9;

volatile unsigned int count = 0;

void test_isr() {
	count++;
}

void setup() {
	Serial.begin(115200);
	while (!Serial);

	if (!dac0.begin(AN_RESOLUTION_12, 8000, 32, 64)) {
		Serial.println("Failed to start DAC1 !");
		return;
	}

	attachInterrupt(digitalPinToInterrupt(ServoPin), test_isr, CHANGE);
}

void loop() {
	const unsigned int c = count;  // read from current, atomic
	static unsigned int prevCount = c;

	if(prevCount != c) {
		Serial.print(c);
		Serial.print('\t');
		Serial.println(micros());
		prevCount = c;
	}
}
```

If that's working as expected, you should see +1 on every signal change, with ~1500 micros between up/down and ~20,000 micros between pulses. The result shouldn't change depending on whether `dac0.begin()` is called or not.

---

Also try removing the `noInterrupts()` / `interrupts()` calls in the `getPulseInternal()` function, see if that is possibly causing issues:

https://github.com/dmadison/ServoInput/blob/f57a6258de51182c4591a931bdd1ead4072ad5fe/src/ServoInput.h#L492-L499

That's necessary on 8-bit boards, but might be causing problems with the Mbed OS.

npf, 2025-01-08:
Running your code with the `if (!dac0.begin(AN_RESOLUTION_12, 8000, 32, 64)) {` commented, I get
```
22:40:23.698 -> 727	7973416
22:40:23.698 -> 728	7974566
22:40:23.698 -> 729	7993418
22:40:23.698 -> 730	7994543
22:40:23.730 -> 731	8013395
22:40:23.730 -> 732	8014520
22:40:23.730 -> 733	8033372
22:40:23.730 -> 734	8034472
22:40:23.762 -> 735	8053351
22:40:23.762 -> 736	8054449
22:40:23.795 -> 737	8073351
22:40:23.795 -> 738	8074427
22:40:23.795 -> 739	8093329
22:40:23.795 -> 740	8094379
22:40:23.826 -> 741	8113306
22:40:23.826 -> 742	8114356
22:40:23.859 -> 743	8133308
```

npf, 2025-01-08:
with the `dac0.begin` block uncommented, I get
```
22:44:24.883 -> 1	4294781249
22:44:24.883 -> 2	4294781249
22:44:24.904 -> 3	4294781248
22:44:24.904 -> 4	4294781249
22:44:24.904 -> 5	4294781248
22:44:24.904 -> 6	4294781249
22:44:24.936 -> 7	4294781248
22:44:24.936 -> 8	4294781249
22:44:24.936 -> 9	4294781248
22:44:24.968 -> 10	4294781249
22:44:24.968 -> 11	4294781249
22:44:24.968 -> 12	4294781249
22:44:25.001 -> 13	4294781249
22:44:25.001 -> 14	4294781249
22:44:25.001 -> 15	4294781249
22:44:25.001 -> 16	4294781249
22:44:25.033 -> 17	4294781249
22:44:25.033 -> 18	4294781249
22:44:25.065 -> 19	4294781249
22:44:25.065 -> 20	4294781249
...
22:44:27.436 -> 258	4294781249
22:44:27.436 -> 259	4294781249
22:44:27.467 -> 260	4294781249
22:44:27.467 -> 261	4294781249
22:44:27.467 -> 262	4294781249
22:44:27.499 -> 263	4294781248
22:44:27.499 -> 264	4294781248
22:44:27.499 -> 265	4294781248
22:44:27.499 -> 266	4294781248
22:44:27.531 -> 267	4294781248
22:44:27.531 -> 268	4294781249
22:44:27.563 -> 269	4294781248
22:44:27.563 -> 270	4294781249
22:44:27.563 -> 271	4294781249
22:44:27.563 -> 272	4294781248
22:44:27.595 -> 273	4294781249
22:44:27.595 -> 274	4294781248
22:44:27.595 -> 275	4294781249
22:44:27.628 -> 276	4294781249
22:44:27.628 -> 277	4294781249
22:44:27.628 -> 278	4294781249
...
```

npf, 2025-01-08:
Each time I reset the Arduino, I get a different set of 2 consecutive values:
```
22:50:47.098 -> 1	4294966834
22:50:47.098 -> 2	4294966834
22:50:47.129 -> 3	4294966835
22:50:47.129 -> 4	4294966834
22:50:47.129 -> 5	4294966835
22:50:47.129 -> 6	4294966834
22:50:47.161 -> 7	4294966835
22:50:47.161 -> 8	4294966834
22:50:47.161 -> 9	4294966835
22:50:47.194 -> 10	4294966834
22:50:47.194 -> 11	4294966834
```
RESET, then:
```
22:52:06.751 -> 1	4294966841
22:52:06.751 -> 2	4294966842
22:52:06.780 -> 3	4294966842
22:52:06.780 -> 4	4294966841
22:52:06.780 -> 5	4294966842
22:52:06.780 -> 6	4294966841
22:52:06.812 -> 7	4294966842
22:52:06.812 -> 8	4294966841
22:52:06.812 -> 9	4294966841
22:52:06.845 -> 10	4294966841
22:52:06.845 -> 11	4294966841
22:52:06.845 -> 12	4294966841
22:52:06.877 -> 13	4294966841
22:52:06.877 -> 14	4294966841
22:52:06.877 -> 15	4294966841
22:52:06.877 -> 16	4294966842
22:52:06.909 -> 17	4294966842
```
aso

npf, 2025-01-08:
back to your first code, with `noInterrupts()` / `interrupts()` commented in `ServoInput.h`, I don't see a change:
```
23:13:05.002 -> 0
23:13:05.002 -> 0
23:13:05.002 -> 0
...
23:13:05.002 -> 0
23:13:05.002 -> 0
23:13:05.038 -> 0
23:13:05.038 -> 0
23:13:05.038 -> 0
23:13:05.038 -> 0
23:13:05.038 -> 0
23:13:05.038 -> 0
23:13:05.038 -> 0
23:13:05.038 -> 0
23:13:05.038 -> 0
23:13:05.039 -> 0
23:13:05.039 -> 0
23:13:05.039 -> 0
23:13:05.039 -> 0
23:13:05.039 -> 0
...
23:13:05.039 -> 0
23:13:05.039 -> 0
23:13:05.039 -> 0
23:13:05.039 -> 1
23:13:05.039 -> 1
23:13:05.039 -> 1
...
23:13:05.077 -> 1
23:13:05.077 -> 1
23:13:05.077 -> 1
23:13:05.077 -> 0
23:13:05.077 -> 0
23:13:05.077 -> 0
...
23:13:05.077 -> 0
23:13:05.077 -> 0
23:13:05.077 -> 0
23:13:05.109 -> 0
23:13:05.109 -> 0
23:13:05.109 -> 0
... (several repetitions of sequences of 0 or 1 values)
23:13:11.891 -> 0
23:13:11.891 -> 0
23:13:11.891 -> 0
23:13:11.891 -> 0
23:13:11.891 -> 0
23:13:11.891 -> 4294967295
23:13:11.891 -> 4294967295
23:13:11.891 -> 4294967295
23:13:11.891 -> 4294967295
23:13:11.891 -> 4294967295
...
23:13:11.983 -> 4294967295
23:13:11.983 -> 4294967295
23:13:11.983 -> 4294967295
23:13:11.983 -> 0
23:13:11.983 -> 0
23:13:11.983 -> 0
...
```

dmadison, 2025-01-09:
This library uses the `micros()` function as a board-agnostic solution for timing interval calculations.  It appears that the `Arduino_AdvancedAnalog` library is interfering with the timer used for the `micros()` function. I would recommend creating an issue on their repo: https://github.com/arduino-libraries/Arduino_AdvancedAnalog/issues

The `micros()` function is provided by the Arduino platform for the Giga board. In case it's helpful for troubleshooting, you can find details on the implementation in that repo here: https://github.com/arduino/ArduinoCore-mbed

Since the problem doesn't seem to be caused by this library I'm going to close this issue and mark it as resolved. Good luck!

closed 2025-01-09 by dmadison